50an6xy06r6n/hotswap_pcb_generator #1: npm install doesn't properly install the kle-serial dependency

50an6xy06r6n, 2021-06-28:
The first time you run `npm install`, it may not run the prepublish script inside of `kle-serial`. This only happens the first time, and even then doesn't always repro. I think this has something to do with the fact that it installs a fork of `kle-serial` directly from GitHub, and the fact that the prepublish script has been deprecated for a while. I think it might be fixable by converting `kle-serial` to use a prepare script, but I'm not sure because I can't repro consistently. If anyone has a good fix for this please let me know.

faster-harder-coder, 2021-06-29:
Is there a workaround for this? Running `npm install` twice does not help. I tried with both node v16.4.0 and v14.0.0.

50an6xy06r6n, 2021-06-29:
Did you run `npm install` inside the `kle-serial` directory (inside `node_modules`)? You can also try running `npm run build` in that directory, which should fix it for sure.

asommer70, 2021-07-18:
On Ubuntu 20.04 doing ```npm install``` inside the kle-serial directory worked for me after doing ```npm install typescript```.  Just FYI in case it helps anyone.

djwink1, 2022-08-30:
Not sure if this if the right section to ask this but are there more detailed or "step by step, start to finish" instructions on how and where to install and run the script for the KLE to SCAD conversion on a windows PC? I'm new to this.

djwink1, 2022-09-12:
Awesome, thanx!!