50an6xy06r6n/hotswap_pcb_generator #4: Switch key RGB light hole

AndyChiu, 2021-07-01:
Hi, is possible add RGB light hole? now have diodes on the way
if have light hole, maybe can add LED RGB strip light easy!

50an6xy06r6n, 2021-07-02:
do you mean holes for the LED wires, or just a through hole for the light to go through. Either way it will be tricky because the diode is in the way for the MX socket. I'll think about it, but since this is primarily intended for prototyping it doesn't seem like a critical feature.

AndyChiu, 2021-07-02:
Hi, just a through hole for the light to go through, i hope it will be used for a long time. If can added backplane, and add config standoff can pass through the PCB, and allow lock screw with the backplane, just like the sandwich structure, it will be used for a long time.

AndyChiu, 2021-07-31:
[image]
Hi, i try modify code to add LED cutout

                // Diode cathode cutout
                translate([3*grid,2.4*grid,0])
                    cylinder(h=pcb_thickness+1,r=.7,center=true);

                // Diode Channel
                translate([0,2.4*grid,pcb_thickness/2])
                    cube([6*grid,1,2],center=true);
                translate([.3*grid-.5,2.4*grid,pcb_thickness/2])
                    cube([4*grid,2,3],center=true);

                // Add LED cutout
                translate([0,-4*grid,pcb_thickness/2])
                    cube([5,4,10],center=true);

the Diode is more easy to install

[image]

renatosuero, 2022-02-03:
Hi, I'm new here, how is the led and the connection at the wire/diode?

AndyChiu, 2022-02-03:
Hi, I'm still think about how to connect , but i think use the LED  strip is be a easy way
[image]

slabua, 2022-04-29:
@AndyChiu Hi, do you have the individual stl for the mx switch with the led hole? I seem I can't find it in your fork, and not quite sure about how to generate it. Thank you.

slabua, 2022-04-29:
@AndyChiu I think I have figured it out. Please enable issues on your fork :)

AndyChiu, 2022-04-29:
hi, i was upload same stl file, and update my .scad, have fun! :)