50an6xy06r6n/hotswap_pcb_generator #7: Add support for solid cases

50an6xy06r6n, 2021-07-20:
I think integrated-plate cases are feasible, and could be a nice feature to have for more permanent solutions.

AndyChiu, 2021-07-21:
Hi, the generator is feasible create like this keyboard case ? have around wall
https://github.com/rtitmuss/torn/blob/master/case/HiPro%20Case/readme.md

50an6xy06r6n, 2021-07-22:
What I'm thinking is less complicated. Would probably be integrated with the plate or backplate

AndyChiu, 2021-07-22:
It's Great!

AndyChiu, 2021-07-22:
This have some idea 
https://github.com/takjn/kbcgen

50an6xy06r6n, 2021-07-23:
Thanks, that gave me some ideas on how to make plate shape generation prettier overall.

50an6xy06r6n, 2021-07-26:
Started working on it in the [solid_case](https://github.com/50an6xy06r6n/hotswap_pcb_generator/tree/solid_case) branch

50an6xy06r6n, 2021-07-27:
Merged solid case code into main and added example STLs

closed 2021-07-27 by 50an6xy06r6n