50an6xy06r6n/hotswap_pcb_generator #12: some components

AndyChiu, 2021-09-12:
Hi, i add some components and use UTP cable to create keyboard, i modified part of the file, and created another folders to separate, the folder name with '_UTP' 
https://github.com/AndyChiu/hotswap_pcb_generator

50an6xy06r6n, 2021-09-27:
Hey sorry I've been pretty busy recently but I will try to take a look at the components you added. Particularly interested in the encoders. In general I think I need to find a nice modular way for people to add new components without major changes to existing code. Not sure exactly how to do that yet.

AndyChiu, 2021-09-27:
Hey, i was write a blog to show how to create keyboard with hotswap_pcb_generator, but now only chinese version
https://ie321mx.blogspot.com/2021/09/choc-v2.html
https://ie321mx.blogspot.com/2021/09/blog-post.html

aamott, 2024-02-27:
I know that this is old, but now that I see a few updates to this repo, I'll add my thoughts. AndyChiu's updates were very nice and I ended up using them for my prototyping, including the LED addition. Updated diode routing was the most helpful change. I'd love to see these features added here!

AndyChiu, 2024-02-27:
Thank you, I will be tidying up my code soon and updating it to my repo. Since I'm currently focusing on designing a curved keyboard, it has diverged a bit from this repository.

AndyChiu, 2024-03-03:
I've updated the repo, I tried to make the MCU more modular, and tried adding holder to the PCB to hold the switch.

https://twitter.com/andychiu/status/1748546787114369427