50an6xy06r6n/hotswap_pcb_generator #15: Question: Can this be used reasonably for making custom non-split boards?

KyleStiers, 2022-02-24:
Hello, I'm interested in using this to make a more standard layout (but still custom) board. I managed to get it installed and to render the correct layout as well as to remove the tent, as it's not a split board, but I can't figure out how to fine-tune the things you put in the design document. I'd like to add the MC and stabilizers and all that so I can test printing it, but any tips on that? 

Do you just change the things in parameters.scad then re-render? The main struggle right now is adding the MCU and getting the whole board to be one pcb piece, as it currently splits it into a few pieces - I think due to spacing being more than some threshold. Also, is there any easy way to split this into two pieces when doing a more standard board like this? My 3d printer won't fit the whole thing, but I could maybe do that after everything else is done in Fusion or something.

I've attached the keyboard layout as a txt file as github isn't allowing JSON.

Thanks,
Kyle

[keyboard-layout.txt](https://github.com/50an6xy06r6n/hotswap_pcb_generator/files/8135474/keyboard-layout.txt)

KyleStiers, 2022-02-24:
[image]

Here is an image of the current keyboard.scad that is generated

AndyChiu, 2022-02-25:
Hi, you can try this file
[layout.scad.txt](https://github.com/50an6xy06r6n/hotswap_pcb_generator/files/8137629/layout.scad.txt)