50an6xy06r6n/hotswap_pcb_generator #16: Backplate not generating correctly

internetrelative, 2022-04-03:
I'm trying to generate the backplate using the jd40 example but I'm not sure what's wrong it

[image]

[scad.zip](https://github.com/50an6xy06r6n/hotswap_pcb_generator/files/8405039/scad.zip)

AndyChiu, 2022-04-03:
Tenting?
[image]

internetrelative, 2022-04-03:
> 
ah that solves it, sorry It's my first time with openscad. How do you generate screw points to mount the backplate to the case of the keyboard?

AndyChiu, 2022-04-04:
hi, you can read this guide, and find Standoff Placement part
https://github.com/50an6xy06r6n/hotswap_pcb_generator/blob/main/guide/design_guide.md

internetrelative, 2022-04-05:
I got it to work :) , didn't know I had to specify the location of the standout manually 

[image]

[jd40_scad.zip](https://github.com/50an6xy06r6n/hotswap_pcb_generator/files/8420636/jd40_scad.zip)

closed 2022-04-05 by internetrelative