50an6xy06r6n/hotswap_pcb_generator #24: How do these affect height / Please provide side view, or schematic

OWL4C, 2024-10-02:
Without building these, i wonder how much they affect the height of a keyswitch. Having made my case in ergogen already, i was hoping to add these to the wall height of my case, but  i do not know how much they change the height of the switch.

From a youtube video, i saw that they do not sit flush with the bottom of the switch, since the diode is between them. Maybe that was only the videos fault- i don't know.
It would be nice if you could take some pictures from the side and / or give some measurements (or ranges, if they can slide up and down).
Really nice would be the amount of space between the casing (lower edge without stem) to the upper pcb, since feet length can vary.
Since different sources have different feet length (between 2-3 mm for the contacts) and the pcb i created is 4mm thick, knowing how near they can get to the switch casing would be enough to plan my case.

50an6xy06r6n, 2024-10-03:
There's a YouTube video about this? That's news to me. Can you link it so I can see what gap you're referring to?

The PCB is designed so that the switches can sit flush when using the glass-encased variety of 1n4148 diodes. I have not tested it using any other diode types. Maybe that should go in the README if it isn't already.

OWL4C, 2024-10-03:
There was this 3 hour livestream which is really more about the youtuber chatting than the project. 
https://www.youtube.com/watch?v=3ZCAY2VO7p8 At around 2:06 you can see the switch get inserted but because of the perspective i couldn't figure out if it is flush or slightly sitting on top of the diode. Looking at it one a bigger screen now you can reasonably suspect they sit flush though.

If they sit flush with the bottom housing, by my own measurements and this website https://media.digikey.com/pdf/Data%20Sheets/Cherry%20PDFs/MX%20Series.pdf you could mention that at 4mm they will increase the height by only .7mm; as for me, i'll just make the case 1mm higher, and the keys should still reach the pcb comfortably. I'll be sure to update this as soon as the print arrives and i can test out the fit!

closed 2024-10-03 by OWL4C

OWL4C, 2024-11-24:
Okay i have tried multiple settings for a Creality Ender 3 V3 Se, but with a 0.4mm default nozzle and many different speeds, i still get unusable prints, without holes or indentations. Are there any recommendations for settings? Maybe even for a Ender Style printer?

50an6xy06r6n, 2024-12-06:
What do you mean without indentations? This sounds more like a print quality/possibly slicer issue tbh. I don't think the choice of printer really matters that much if it's dialed in (I printed mine off of a self-sourced Kossel mini with questionable calibration).

Also, as noted in the documentation, you can use a tiny drill bit and pin vise to open up holes that the printer doesn't print properly. It's tricky since the holes are sized for a snug fit, so you can't oversize them as much as you would for other applications.

50an6xy06r6n, 2024-12-06:
Would definitely recommend printing individual components to test calibration and stuff. Worst case, you can increase the hole sizing locally (I think most of those things are configurable in `parameters.scad`).

OWL4C, 2024-12-18:
[image]
[image]
this one is one of the best attempts. Besides the channels being clogged, which can easily be drilled, the top is also quite rough and melted, probably some temp problem? 
The main issue is the missing holes for the switch pins, and the missing upper channel (non-diode leg). I am quite new to 3d printing and have no idea how to fix this, except use a .2 nozzle and more ventilation / cooling.

OWL4C, 2024-12-18:
[image]
Actually it seems like the mesh was broken from the start...
At least compared to the layout guides images.

AndyChiu, 2024-12-18:
Hello, could you please provide layout.scad and parameters.scad? This should provide a common basis for checking the problem.

50an6xy06r6n, 2024-12-18:
Yeah that model definitely doesn't look right. What does it look like in OpenSCAD? It looks like you're potentially combining some settings for row- and column-oriented layouts that aren't compatible.

On the fabrication side, it looks like your print has some pretty severe overextrusion, which would cause holes to be undersized and/or close up

OWL4C, 2024-12-18:
(One of) My layout: https://github.com/OWL4C/hotswap_pcb_generator/blob/main/l_layout.scad
And yes same geometry in scad: 
[image]

50an6xy06r6n, 2024-12-18:
Hmm there's nothing obviously wrong with the layout there... I will have to take a look later

OWL4C, 2024-12-18:
Thanks a bunch! I'll make sure to push my entire working folder with all the files to that repo, maybe something else is wrong.